momentoscope/hextof-processor #3: Clean up readme and generate documentation

balerion, 2018-04-30:
We need to do some serious cleanup regarding the readme and creating the documentation. Examples need to be moved to comments in the code, and the documentation page needs to be generated.

RealPolitiX, 2018-05-07:
The first version of the documentation is built as html (you can pull the current version and view the **index.html** in browser), just looking for a good host such that it can be accessed online.

balerion, 2018-05-07:
what about https://pages.github.com
?

RealPolitiX, 2018-05-10:
I used sphinx for doc generation, which gave no problems, but the readthedocs website can't rebuild it well, http://hextofofflineanalyzer.readthedocs.io/en/latest/
It's missing texts present in the docs folder which I think is because this is not yet an official installable package
Github pages use jekyll to render, but it seems that it's no trivial task to get it working with github pages and jekyll, https://jefflirion.github.io/sphinx-github-pages.html
the current status is like this...
https://momentoscope.github.io/HextofOfflineAnalyzer/

RealPolitiX, 2018-05-10:
I'll try to implement the approach from https://jefflirion.github.io/sphinx-github-pages.html

RealPolitiX, 2018-05-10:
Great, it's working now!

RealPolitiX, 2018-05-11:
@balerion @steinnymir The latest version of documentation is uploaded. I have incorporated most of the readme into the formal documentation. In principle we can delete point 1-4 in the README.md
@balerion Is point 5 in the README.md the same as in the utils docstring?

RealPolitiX, 2021-01-15:
Close this as recent changes have been mirrored in the documentation.

closed 2021-01-15 by RealPolitiX