momentoscope/hextof-processor #16: Reporting of electron fraction in binned volume

momentoscopists, 2018-05-02:
It would be nice if the binner function would return in the end a report of the fraction of electrons in the binned volume (e.g. 15 Mcts out of a total of 20 Mcts in the run binned (75%))

steinnymir, 2018-05-07:
I will add such information to the calcculateBinnedArray function, on which I am working already for the h5 output format.

balerion, 2020-12-10:
This would require, among other things, to include self.numOfElectrons in the parquet files. I think this would be a really useful upgrade

steinnymir, 2020-12-10:
I agree, and this recalls to the general issue of populating metadata in the process of generating the dataframe. This is partially implemented but need some more work to get it up and running.