momentoscope/hextof-processor #19: Optionalize ddMicrobunches

balerion, 2018-05-03:
I have been thinking of making the whole ddMicrobunches dataframe optional. Is this even necessary at all? we should have all the data in the dd dataframe anyway. If needed for debugging purposes, we can generate it optionally. Or, we could always generate it but have it empty if not required?

Just a proposition, no need to fuss about this if not necessary.

steinnymir, 2018-05-07:
I totally agree with this. The generation of ddMicrobunch is time and space consuming, while almost never used at all.

balerion, 2018-05-08:
The ddMicrobunches is only used for diagnostics and for generating the histograms. We are looking into generating the histograms in other ways, using the dd dataframe, so this will be optionalized.