momentoscope/hextof-processor #21: Avoid progress bar spam

steinnymir, 2018-05-08:
when running the code outside of IPython, the progress bar is not updating itself and it is rather spamming a new line for each iteration.

During the dataframe creation, the first iteration takes far longer (5 minutes) than all other iterations, (10 seconds max), and also, the progress bar spams a line each 0.1 seconds, instead of at each cycle.

I dont know whether this is a bug or a setting which can be set, but it would be nice, if it cannot be repaired, to set the use of the progress bar optional.

RealPolitiX, 2018-05-08:
By "spam" you mean "span"? The progress bar for distributed computing is a bit tricky, but dask has its own, which only display the progress of .compute() method

steinnymir, 2018-05-08:
I ment "spam" in the sense that for me it prints a new line for each iteration (each time stamp in the with progressbar(): statement.

Did some digging and found this is not the fault of the progress bar, but of PyCharm, which doesnt support the backspace character.

It would be great therefore to be able to choose between the old method and the progress bar.

steinnymir, 2018-05-08:
I already fixed this with an option for turning this progress bar off in the dataframe creation section. The binning part is less annoying since it is only about 10 lines or so.

RealPolitiX, 2018-05-10:
Alright, it's really strange, cuz I tested things on jupyter notebook before committing, there was no such problems.

steinnymir, 2018-05-10:
problems come from PyCharm's lak of backspace character, it works everywhere else.

closed 2020-11-23 by steinnymir