momentoscope/hextof-processor #29: KeyError: 'monochromatorEnergy'

zainsohail04, 2020-11-26:
In the calibrate function, there's a useAvgMonochormatorEnergy parameter. Whether it's false or true, both ways it gives the error KeyError: 'monochromatorEnergy'. I saw that it's in the TODO but I couldn't figure out a way to bypass that in the function.

balerion, 2020-11-29:
this should be fixed by #36
Let me know if it works for you!

zainsohail04, 2020-11-29:
It isn't working. I even did a fresh git checkout but I get this error:
`KeyError                                  Traceback (most recent call last)
~/.conda/envs/hextof-env/lib/python3.7/site-packages/pandas/core/indexes/base.py in get_loc(self, key, method, tolerance)
   2894             try:
-> 2895                 return self._engine.get_loc(casted_key)
   2896             except KeyError as err:

pandas/_libs/index.pyx in pandas._libs.index.IndexEngine.get_loc()

pandas/_libs/index.pyx in pandas._libs.index.IndexEngine.get_loc()

pandas/_libs/hashtable_class_helper.pxi in pandas._libs.hashtable.PyObjectHashTable.get_item()

pandas/_libs/hashtable_class_helper.pxi in pandas._libs.hashtable.PyObjectHashTable.get_item()

KeyError: 'monochromatorPhotonEnergy'

The above exception was the direct cause of the following exception:

KeyError                                  Traceback (most recent call last)
<ipython-input-8-27287a8c69b2> in <module>
      1 prc.calibrateEnergy(toffset=355.,eoffset=113.29,l=0.8, # parameters defined by microscope settings. See documentation for details.
----> 2     applyJitter=True,jitterAmplitude=1,jitterType='normal')# jitter parameters used to remove artifacts.

~/hextof-processor/processor/DldProcessor.py in calibrateEnergy(self, toffset, eoffset, l, useAvgSampleBias, k_shift_func, k_shift_parameters, applyJitter, jitterAmplitude, jitterType, useAvgMonochormatorEnergy, useAvgToFEnergy)
    575 
    576         if useAvgMonochormatorEnergy:        # TODO: add monocrhomator position,
--> 577             eoffset += np.nanmean(self.dd['monochromatorPhotonEnergy'].values)
    578         else:
    579             eoffset += self.dd['monochromatorPhotonEnergy']

~/.conda/envs/hextof-env/lib/python3.7/site-packages/dask/dataframe/core.py in __getitem__(self, key)
   3535 
   3536             # error is raised from pandas
-> 3537             meta = self._meta[_extract_meta(key)]
   3538             dsk = partitionwise_graph(operator.getitem, name, self, key)
   3539             graph = HighLevelGraph.from_collections(name, dsk, dependencies=[self])

~/.conda/envs/hextof-env/lib/python3.7/site-packages/pandas/core/frame.py in __getitem__(self, key)
   2900             if self.columns.nlevels > 1:
   2901                 return self._getitem_multilevel(key)
-> 2902             indexer = self.columns.get_loc(key)
   2903             if is_integer(indexer):
   2904                 indexer = [indexer]

~/.conda/envs/hextof-env/lib/python3.7/site-packages/pandas/core/indexes/base.py in get_loc(self, key, method, tolerance)
   2895                 return self._engine.get_loc(casted_key)
   2896             except KeyError as err:
-> 2897                 raise KeyError(key) from err
   2898 
   2899         if tolerance is not None:

KeyError: 'monochromatorPhotonEnergy'`

steinnymir, 2020-11-29:
it seems like you might not have the mono channel in... could you post the code that generated this? And the DAQ_channels section in SETTINGS.ini

zainsohail04, 2020-11-29:
I am trying to run the "Tutorial_01_Binning..."

These are the DAQ channels generated by running python InitializeSettings.py:

[DAQ channels]
dld_pos_x = /uncategorised/FLASH1_USER2/FLASH.FEL/HEXTOF.DAQ/DLD1:0/dset
dld_pos_y = /uncategorised/FLASH1_USER2/FLASH.FEL/HEXTOF.DAQ/DLD1:1/dset
dld_time = /uncategorised/FLASH1_USER2/FLASH.FEL/HEXTOF.DAQ/DLD1:3/dset
dld_detector_id = /uncategorised/FLASH1_USER2/FLASH.FEL/HEXTOF.DAQ/DLD1:3/dset
dld_microbunch_id = /uncategorised/FLASH1_USER2/FLASH.FEL/HEXTOF.DAQ/DLD1:2/dset
dld_aux_0 = /uncategorised/FLASH1_USER2/FLASH.FEL/HEXTOF.DAQ/DLD1:4/dset
dld_aux_1 = /uncategorised/FLASH1_USER2/FLASH.FEL/HEXTOF.DAQ/DLD1:4/dset
delay_stage = /Experiment/Pump probe laser/delay line IK220.0/ENC
bam = /Electron Diagnostic/BAM/4DBC3/electron bunch arrival time (low charge)
bunch_charge = /Electron Diagnostic/Bunch charge/after undulator
macro_bunch_pulse_id = /Timing/Bunch train info/index 1.sts
optical_diode = /Experiment/PG/SIS8300 100MHz ADC/CH9/pulse energy/TD
gmd_tunnel = /Photon Diagnostic/GMD/Pulse resolved energy/energy tunnel
gmd_bda = /Photon Diagnostic/GMD/Pulse resolved energy/energy BDA
pump_pol = /uncategorised/FLASH1_USER2/FLASH.EXP/NF.ESP301/PG2/MOTOR3.POS/dset

steinnymir, 2020-11-29:
The problem is you are not reading in the data for the monochromator photon energy.
Try adding this: `monochromator_photon_energy` = /FL1/Beamlines/PG/Monochromator/monochromator photon energy` together with the other channels. I can't say it will work for sure, since I don't know if the channel was the same in that beamtime..

zainsohail04, 2020-11-30:
You were right about that. But I found the path in the hdf5 file to be "/Beamlines/PG/Monochromator/monochromator photon energy" for monochromator. Should I commit to make the tutorial.ini compatible with the tutorial?

steinnymir, 2020-11-30:
Yes!

zainsohail04, 2020-12-04:
Fixed in #40

closed 2020-12-04 by zainsohail04