momentoscope/hextof-processor #61: Enhancement needed when storing dataframes loaded with id interval

balerion, 2020-12-22:
The metadata gets overwritten by the last interval when storing dataframes in append mode. In other words, the data gets appended, but the metadata is only aware of the last append.

steinnymir, 2021-04-21:
this has been fixed in this [commit](https://github.com/momentoscope/hextof-processor/commit/e866838f62c95d2cbdf39728143e7f6030541649) in the hackathon branch.

now metadata is correctly retained when appending parquets. The json metadata file for the appended parquets contains a dictionary with an aggregate of the runInfo of all included runs, to represent correctly the number of electrons/macrobunches, run numbers etc. The metadata of each run is stored next to this as dictionaries with the same structure as before.

zainsohail04, 2021-09-29:
Should we close this issue if this has been fixed?

steinnymir, 2021-09-30:
yes, should be fixed! 

If there is any problem with metadata handling, we should make a new issue

closed 2021-09-30 by steinnymir