momentoscope/hextof-processor #62: Major bug in PAH

balerion, 2020-12-23:
There is a major bug in the PAH module (inside h5filedataaccess.valuesFromInterval) that results in ~8% of the electrons being lost, mostly at the end of the macrobunch.
The h5 files come in chunks of ~3200 macrobunches, and each chunk has its own shape. However, all the chunks are loaded as if their shape was the same as the first one, resulting in cutting most other chunks significantly. This severely impacts the unpumped data, usually at the end of macrobunches.

balerion, 2020-12-23:
For now, my suggested workaround is to generate parquet files by loading runs in small pulse id intervals (300 macrobunches) to limit the lost electrons. This is a lengthy process, but only needs to be done once.

RealPolitiX, 2020-12-28:
This is mainly related to another package, so maybe should be posted on the bitbucket [repo](https://stash.desy.de/projects/CS/repos/pah/browse) of ``pah``. Alternatively, you can consider forking the ``pah`` package under momentoscope and do monkey patching.

steinnymir, 2020-12-28:
The best solution to this problem is to get rid of the PAH dependency, which is planned anyways for the new coming data structure. See issue #56

balerion, 2021-01-07:
@steinnymir True, but this severely limits the usability of older datasets where only the old file structure is available. Perhaps I will try following @RealPolitiX's suggestion

steinnymir, 2021-01-07:
Not necessarely. The new readout system can be adapted for the old data structure, this should not be too complex. It would even better allow us to understand the changes this brings, by reading the old data with both methods and comparing, isolating problems which might arise from the new data structure itself.

balerion, 2021-01-08:
@steinnymir I followed your suggestion, now all the electrons are loaded. This comes with the side effect that for large datasets, a lot of memory is required, often causing kernel crashes if loading the whole dataset in memory (will open new issue)

closed 2021-01-08 by balerion