momentoscope/hextof-processor #70: Delete all branches except master for LFS integration

zainsohail04, 2021-01-20:
I deleted the branches I had on the repo and I believe Steinn deleted some too. There's still two branches here. Is it possible to delete them?

steinnymir, 2021-01-20:
Today I also deleted all branches which were already merged, or were too old and did not contain useful additions . I wrote to @Jus80687 to ask about the branch Scholz2018 as it was created for one of his beamtimes. If he gives the go, you can close it .

The last one is the work @balerion was doing on the orthoslices visualizer. I don't know the state of it, but maybe we could copy th changes and create a new branch with these after the LFS fix?

steinnymir, 2021-01-20:
The Scholz2018 branch is gone, last one standing is the orthoslices one, which is up to @balerion to decide how to handle it (merge or delete).

balerion, 2021-01-21:
It's gone

zainsohail04, 2021-01-21:
The task was done

closed 2021-01-21 by zainsohail04