momentoscope/hextof-processor #71: LFS quota

balerion, 2021-01-21:
So there appears to be an LFS 1Gb quota:

Bandwidth quota
If you use more than 1 GB of bandwidth per month without purchasing a data pack, Git LFS support is disabled on your account until the next month.

Which means, right now I cannot even LFS fetch. Is this the right approach after all?

balerion, 2021-01-21:
The smallest datapack is 5$/month, for 50Gb. This would solve the issue imo, but we need to agree on payment resposibility. My uni has a weird policy about not paying monthly subscriptions, but I can try to push a little.

zainsohail04, 2021-01-21:
Is it so that you currently can't pull those files? I saw this before but I was of opinion they are not enforcing this limit for open repositories.

zainsohail04, 2021-01-21:
[FLASH DAQ HDF](https://gitlab.desy.de/christopher.passow/flash-daq-hdf/-/tree/master/) repository also makes use of git LFS. They are using desy hosted gitlab but perhaps we can take assistance from them and store the LFS objects on DESY servers, rather than github.

RealPolitiX, 2021-01-21:
OK, in any case, there could be a [hextof-data](https://github.com/momentoscope/hextof-data) repo. If you decide to use DESY service in the end, you can list the datasets on ``hextof-data`` as links, this also applies to any dataset deposited to Zenodo. I think for storing calibration datasets, hextof-data is sufficient.

zainsohail04, 2021-01-21:
> OK, in any case, there could be a [hextof-data](https://github.com/momentoscope/hextof-data) repo. If you decide to use DESY service in the end, you can list the datasets on `hextof-data` as links, [...]

It was challenging to move the LFS files from github to desy gitlab but now I did manage to do it (in a [hextof-data repo](https://gitlab.desy.de/momentoscope/hextof-data) there). 
Please check if forking the data works as expected and suggest how to continue. 
Shall I remove the data files from the repository here? 
Please also give me usernames for accounts which need ownership of that repository.

steinnymir, 2021-01-22:
I cannot log in to the desy GitLab with my credentials, so I cannot be owner of that one. It might be that it is closed to DESY members. 

Did the LFS cleanup work? if so, we don't *need* to delete the data from hextof-processor, but it is also useless to have it duplicated in the different repos. I would remove it... @balerion @RealPolitiX what do you think?

zainsohail04, 2021-01-22:
In my earlier reply here, I mentioned that I can not push changes (for LFS cleanup) due to quota restrictions. We'd need to buy a plan or wait till it resets. I thought using DESY gitlab for data makes sense since I didn't see any quota restrictions etc. there.

steinnymir, 2021-01-22:
Sorry, I missed that. I agree that using the DESY GitLab for LFS is a better option, as licenses might be a problem to finance (even when cheap..).

If the data is available on the GitLab repository, and even mirrored in a github one through links (I feel its redundant though), I see no reason to keep it in the hextof-processor, so if possible I would just remove it all. If however the LFS quota blocks us from doing so, lets set a reminder to do this once we have quotas again.

balerion, 2021-01-25:
@steinnymir #issuecomment-765343803
Sure, I think it is a good idea

zainsohail04, 2021-02-15:
I am unclear why the quota hasn't reset. However, we should remove the LFS files so that commits can be published to the repository. Should I contact GitHub support, or would someone else do it?

steinnymir, 2021-02-15:
Yes, you can do it. I don't understand why it didnt unlock, but I guess if you explain them we need the quota unlocked to remove usage of this feature, it should not be too much of a problem.

zainsohail04, 2021-02-17:
They reset the quota so I used bfg to clean the repo from .h5 files, and also remove the Git LFS integration. The repo should now be clear of data files. If you can confirm this, please close this issue.

steinnymir, 2021-02-17:
Cloning a new copy of the repo results in a 37MB folder now. And I see no trace of LFS. Thanks for solving this, I will close the issue now.

closed 2021-02-17 by steinnymir