momentoscope/hextof-processor #72: Error reading settings files when package installed via pypi

zainsohail04, 2021-01-26:
The DldProcessor Class assumes a root folder of 
`root_folder = os.path.dirname(os.path.dirname(processor.__file__))`

However, if one installs the package in a certain environment, this would lead to it searching in an incorrect folder. Perhaps this should be user customisable in that case? or is there already a way?

steinnymir, 2021-01-26:
The "root_folder" defined as it is there, should work anywhere you install your package, as it looks for the imported *processor* and goes up one folder. It will be easy to see if this breaks however, as the whole settings (required for almost every step) depends on it...

zainsohail04, 2021-01-26:
If I look in my conda env location
`cd .conda/envs/<MY_ENV>/lib/python3.7/site-packages/processor`
I am listed these files/dirs
`cscripts  DldFlashDataframeCreator.py  DldProcessor.py  __init__.py  legacy  pah.py  __pycache__  utilities`

There is no settings folder, which explains why I get the exception `Exception: Failed loading main settings!`

steinnymir, 2021-01-26:
Packaging only what is in the processor folder will not work. The way it works now, is that there are settings stored in the settings folder, and these are necessary, but I agree not exactly "package" material. 

Do you have a SETTINGS.ini somewhere? it usually appears in the root folder of the repo, but I don't know what happens with the packaged version... If also this is not possible, we might need to restructure the whole settings system.

zainsohail04, 2021-01-27:
> Packaging only what is in the processor folder will not work. The way it works now, is that there are settings stored in the settings folder, and these are necessary, but I agree not exactly [...]

Yes I understand but since every folder becomes it's own package (e.g. import processor), then one would have a settings package, which doesn't make sense to have as package.

Settings.ini is not initialized when I create an object of the class. I am guessing that's due to the directory being restricted.
Perhaps we could allow the user to define a settings folder path for the pip installed hextof-processor

zainsohail04, 2021-04-18:
We can come back to this once we want to package again.

closed 2021-04-18 by zainsohail04