momentoscope/hextof-processor #75: Binning performance

steinnymir, 2021-04-21:
From the discussion in the issue #73 I showed the binning performance as function of number of cores used, which showed a poor performance with increased number of cores used. This turns out to be wrong however, as this was tested on dataframes with too few electrons.

Working with larger datasets, in the order of 200M electrons, the binning performance greatly improves, up to more than 40 cores at least. It showed, in this case, a speedup of a factor 4 at least between using 4 and 32 cores. 94 cores seemed slightly faster, but I did not make a quantitative study of this.

This should be further investigated, looking for a sweet spot for the number of cores, or a variable number of cores, based on the size of the dataset could be a cool alternative.

zainsohail04, 2021-04-22:
I think that in such a case, an adaptive approach to the number of cores makes most sense. Your testing approach can be used but for a few different sized datasets, and we can interpolate a num_cores depending on dataset size then.

zainsohail04, 2021-08-05:
With the current data structure, the binning performance has vastly improved. Henceforth, I don't think we need to look at improving it further. If you think otherwise, please reopen this issue.

closed 2021-08-05 by zainsohail04