momentoscope/hextof-processor #88: read in of some runs are not working with bugfixes code

MichaelHeber, 2021-09-20:
BT:'/asap3/fs-flash-o/gpfs/hextof/2021/data/11010854/ Run: 39386 (PEEM Chessy)
the read in of at least parquet file2 is at least not working.

BT:'2021 : 11009996' Run: 40093
The readin of at least one Hdf5 file is not working, KeyError: 'Unable to open object (component not found)'

zainsohail04, 2021-09-21:
> BT:'/asap3/fs-flash-o/gpfs/hextof/2021/data/11010854/ Run: 39386 (PEEM Chessy)
> the read in of at least parquet file2 is at least not working.
 Is there some error or it just doesn't read that file?

> BT:'2021 : 11009996' Run: 40093
> The readin of at least one Hdf5 file is not working, KeyError: 'Unable to open object (component not found)'
This error comes as some channel missing from that hdf5 file, I believe.

MichaelHeber, 2021-09-24:
the error for the read-in of the parquet file2 is a
ParquetException: Metadata parse failed: /asap3/fs-flash-o/gpfs/hextof/2021/data/11010854/processed/parquet/per_file/FLASH1_USER3_stream_2_run39386_file2_20210817T093440.1

It accrue somewhere in the exception handling.
last Line of the hextof processor is
L412             self.dfs = [dd.read_parquet(fn) for fn in self.prq_names]

closed 2021-11-03 by zainsohail04