momentoscope/hextof-processor #113: Problem with reading one of lab files

kutnyakhov, 2022-02-17:
I was able to read '/asap3/fs-flash-o/gpfs/hextof/2022/data/11013726/raw/20220114/2.h5', but if I am trying to read '/asap3/fs-flash-o/gpfs/hextof/2022/data/11013726/raw/20220114/3.h5' code is hanging.
Checking that file with simple with h5py.File readout, error appeared, but including (libver='latest', swmr=True) solve the error
[image]

steinnymir, 2022-02-18:
From reading the error message, the hdf5 file was "open for write". 

E.g. 
`f = h5py.file(fname,"w")` 
not followed by 
`f.close()`

The argument swmr kind of goes around this . It stands for "Single Write Multiple Read". 
This allows you to read files which are open for read or write in different processes.

Can you try check where this file might have been open?

kutnyakhov, 2022-02-18:
Do you mean that it is still open for writing by LabDAQ or if it is open in some other notebook?
This is for sure not a case, as some other files were created after this one.
Why then it was possible to read within the next cell with "swmr=True", the beginning of which you see also on the screenshot?

steinnymir, 2022-02-18:
I cannot tell where it is open for write, but thats what the error sais.
Might be in the LabDAQ, as I doubt you would load the data in a notebook using write permissions ("w" instead of "r").

The reason swmr=True makes it work, is that the writing procedure in hdf5 can be set to accept one writer and multiple simultaneous readers. The writing enabled this, and therefore, even though the file is kept open to wrtie by an other process, you can still read it from as many processes as you want.

kutnyakhov, 2022-08-22:
Due to change in file naming this issue is no longer relevant.

closed 2022-08-22 by kutnyakhov