momentoscope/hextof-processor #115: Error while using fixLabFilenameReadout branch

kutnyakhov, 2022-08-22:
While trying to use fixLabFilenameReadout branch with the same notebook used in master branch, error "Too many parameters for typing.Sequence; actual 2, expected 1" appeared
[image]

steinnymir, 2022-08-22:
I am not sure but my first guess is to ask: which python version were you using? 3.8? I think there might be differences in the type hints between versions. @zainsohail04 maybe you know more, as you wrote this part of the code.

kutnyakhov, 2022-08-22:
jhub.desy provides only 3.8 up to now.
In my case, it is Python 3.8.12

kutnyakhov, 2022-08-22:
This line makes an error:
from processor.LabDataframeCreator import LabDataframeCreator

zainsohail04, 2022-08-22:
> @zainsohail04 maybe you know more, as you wrote this part of the code.

I just double checked but actually I didn't put these type hints. They only exist in hextofloader. Perhaps you are using some extension that automatically adds them?

> This line makes an error: from processor.LabDataframeCreator import LabDataframeCreator

I see. It is definitely a internal problem. I will try to see but I need to first understand the logic. Perhaps Steinn comes up with the fix in the meantime.

steinnymir, 2022-08-22:
I looked at this too fast. Its in the functions I just added, and the error is clear.

I fixed it (I hope) and pushed to the branch. You can test this now.

kutnyakhov, 2022-08-22:
First error now fixed, but I've got another error 
AttributeError: 'LabDataframeCreator' object has no attribute 'path'
while executing line
prc = LabDataframeCreator(settings=settings,channels = channels)
[image]

steinnymir, 2022-08-23:
I fixed most of the bugs in the new functions. Now the new error is related to incorrect settings, which I am not sure I can fix:
```
Reading Files /asap3/fs-flash-o/gpfs/hextof/2022/data/11015726/raw/20220808_14h31m00s927_000047 to /asap3/fs-flash-o/gpfs/hextof/2022/data/11015726/raw/20220808_14h31m00s927_000047: 1 new files of 1 total.
ERROR: skipped channels missing in h5 file: ['Actuators/hextof/logic/kmic1/Sample_VUSet', 'SlowData/hextof/logic/kmic1/Sample_VURead', 'SlowData/exposure time']
```

@kutnyakhov  can you test it on your side with your settings?

kutnyakhov, 2022-08-23:
@steinnymir Did you push recent changes or is it still the same as yesterday evening?
I'm still having an error while calling
prc = LabDataframeCreator(settings=settings,channels = channels)

steinnymir, 2022-08-23:
Authentication error blocked my push. Now its up to date! sorry about that

kutnyakhov, 2022-08-23:
While using readRuns I'm getting an error
invalid literal for int() with base 10: 'parquet'
[image]

kutnyakhov, 2022-08-23:
While using readFiles error is
files_to_read must contain Paths
[image]

steinnymir, 2022-08-23:
Pushed a fix to these.

steinnymir, 2022-08-23:
This back and forth is not really effective... can you send me your settings and channels file so I can test it here? I have really little time left now, as I am going on vacation this evening. I will be back online in one week.

kutnyakhov, 2022-08-23:
@steinnymir I guess now both (readFiles and readRuns) are working :)
Thanks a lot for the help and have a nice vacation ;)

steinnymir, 2022-08-23:
great to hear. If this is the case, i'll merge the changes to main.

PR #116 (2022-08-23): Fix lab filename readout
fixes #115
closed 2022-08-23 by steinnymir