momentoscope/hextof-processor #118: Issue with Data read in

MichaelHeber, 2022-10-11:
Reading in run 39190 show some strange values in the monochromator position. Three times the for a short period of time the values are much lower then expected. There are 4 raw files which contain in total 40 entries with all having the same value of 593.023.

prc = DldFlashProcessorExpress(39190 ,settings=settings,daq="fl1user3")
prc.readData(daq="fl1user3")
eph=prc.dd.monochromatorPhotonEnergy.compute()
plt.plot(eph.values)

[image]

kutnyakhov, 2023-02-16:
Checked the same run with SED, but problem is still exist although it looks slightly different in respect to max and min in y axis
[image]

balerion, 2023-02-17:
Could you give me access to the data? i am testing an updated version (on the retrocompatibility branch) fixing a few bugs, and would like to check if this is fixed